IRNAS/ToslinkCNC #1: why  optical link?!

mkeyno, 2016-03-19:
hi 
why do you use the complex  optical link  where as there is couple of robust wireless links such as old one Bluetooth and new one ESP8266,?also there is couple of modify Marlin for laser and plasma cutting  
BR

SloMusti, 2016-03-19:
Plasma cutters and laser cutters generate high voltage and a lot of RF noise, thus jamming Bluetooth or WiFi, which is not great anyhow for robust and reliable real-time communication, which is required on CNC machine to have all motors synchronised.

mkeyno, 2016-03-19:
hi@SloMusti , good experience, I didn't know , however I make laser machine with 120 watt laser tube and HC-06 for witless link connection  , till now its work perfectly , but for plasma I don't have any experience, actually I've plan to use modified Marlin in my next  plasma machine and find many cool guys here make very good wireless serial bridge , do you think is there way to keep the noise away from the broad? the ESP module so tempting even @openhardwarecoza make leserweb interface can be embedded in the ESP flash, so you need nothing just connect the ESP

openhardwarecoza, 2016-03-19:
Ps you should check my progress on Laserweb2.0 (: i built a plasma too,
running smoothieware. Busy adding toolpath offset to laserweb2 to
compensate for the kerf on my plasma

openhardwarecoza, 2016-03-19:
See https://plus.google.com/collection/UbVXZB

openhardwarecoza, 2016-03-19:
Sorry,  more direct link:
https://plus.google.com/+PetervanderWalt/posts/Mv5Je3um89u

SloMusti, 2016-03-19:
@openhardwarecoza  LaserWeb looks great, we will definitely try it out with ToslinkCNC and Arduino running Grbl or Marlin, we were after solution like yours for a while.

Bluetooth/WiFi is an option, however having it operate reliably in workshop environments for real-time transmission can be a problem. Note that the wireless bridge between computer and controller is ok, because communication is buffered at the controller. Toslink CNC is different however, since it is the connection between the controller and stepper drivers themselves. Thus we can have the controller sufficiently far away from the machine. To that point you can have also a wireless RF link is need be, but might as well have it connected to the computer.

openhardwarecoza, 2016-03-19:
I like the optical idea.  (i have a 40a GF start chinese plasma)

openhardwarecoza, 2016-03-19:
Gf = hf

SloMusti, 2016-03-19:
You are definitely in need of an optical system or serious shielding. We had even 3D printers running Marlin about 5m from plasma arc failding from its noise and have since purchased a Telwin cutter, that has internal start in the cutting head.

openhardwarecoza, 2016-03-19:
It's not quite that bad.  Using drag cut for the start cuts down a lot of
the HF noise.  And i have success with the machine 1.5m away from my
laptop.  But i do admire this project's idea.  Any hack that reuses off the
shelf products gets my vote (easy to source)

Video (old,  now a days i cut a lot closer to the material) :
https://youtu.be/0ZMn9tS2_II

My bigger gripe is to get smoothieware to pause and wait for an Arc OK
(building a thc/arc ok is easy,  interfacing to mach is easy.  Interfacing
thc/arc ok to one  og the firmwares,  not easy.

SloMusti, 2016-03-19:
Sure, under normal cut and drag start is ok. As soon as you screw up something or cut across a hole, the problem begins. This could be hacked by modifying plasma cutter to have HF active only on trigger for a few seconds. But generally I dont want to have people digging up cutters.

We have went as far as to have an opto triggered relay at plasma-cutter that triggers it and an experimental system for torch height control, where an LED is powered via a resistor divider from the arc voltage itself, coupled into fibre and then optical amplitude measured far away. Works reasonably well but needs some more playing with. Would be nice if Marlin had a THC software module and just used an ADC for input with a fibre measurement system like this or using an i2c ambient light sensor to read the voltage.

There are a few more options how to go about this,

SloMusti, 2016-03-19:
@openhardwarecoza Could you maybe use some pause signal or something like that. Or reusing rpm control if any is present, since spindles take some time to ramp up and some firmwares support it. Not sure if Marlin or similar does.

openhardwarecoza, 2016-03-19:
We are working on adding it to smoothie.  I am avoiding marlin like the plague.  Just too slow, and the codebase too messy

openhardwarecoza, 2016-03-19:
Reusing the temperature wait: https://github.com/Smoothieware/Smoothieware/issues/841

mkeyno, 2016-03-19:
hi @SloMusti , there is couple of guys such as @HakanBastedt who's working new version of marlin for cutting machine like laser , I discuss with him about how to use special pin state to check readiness of tools (in this case plasma torch) before any command process . I saw in many industrial plasma & laser  machine tools readiness, ( in laser auto focus system and in plasma THC system ) work independently, whereas in Marlin , it is possible to bind   this feature in loop function, I believe one of the big missing part of many open source is to presume the tools is always ready , I hope @openhardwarecoza peter could spare more time for such project

SloMusti, 2016-03-19:
@mkeyno agreed on that, we see this need as well and would be nice to figure out the workflow for open hardware/software machines such they are simple and reliable enough for all use-cases and sufficiently user friendly. Currently we are working towards completing the http://goodenoughcnc.eu/ range of machines and the organized and simple control documentation will then be the missing part.

mkeyno, 2016-03-19:
good luck for that @SloMusti ,I'm following you and I suggest take time to review guys like @openhardwarecoza  and @luc-github work , they are making the last chain of integrated software/hardware embedded device for CNC machine , just connect the machine and start CAD-CAM and printing