IRNAS/ToslinkCNC #2: Clone axis or multiple axis.

Jmjcoke, 2016-04-27:
Hello,

I have built a machine that has X1-X2 and Y1-Y2 then Z. Instead of cloning can I just add rX card with same address for X1-X2 and Y1-Y2? How much trouble will it be to get the limit switches through the toslink? How do I add needed hardware to make switches work?

SloMusti, 2016-04-27:
You can have as many RX cards (Receiver-Transceiver) with the same address as you need, so your setup is perfectly feasible. 

There is support for limit switches, every RX card has one input (limit) and one output (spindle enable or similar). Note that we have implemented this features just recently and will be releasing the new versions after testing in a week or two.

SloMusti, 2016-05-23:
@Jmjcoke please check out the new version now