IRNAS/ToslinkCNC #5: Axis other than xyz?

Jmjcoke, 2016-05-24:
Can I route x,y,z thru toslink and then route an extruder axis directly to the stepper drive or will there be axis synchronization problems?

SloMusti, 2016-05-24:
You can route the extruder independently of toslink, there will be a slight 5us delay, but that generally should have no effect on the machine operation at a hobby scale.

Jmjcoke, 2016-05-24:
Hobby scale? I am building a printer with a 1500mm Y axis, 800mm dual X
carriage, 900mm Z axis. I am going to wait until the firmware for a stacked
tx card is done. I think I can get the second X axis to run thru the second
card if the timing between the cards is synced.

SloMusti, 2016-05-24:
For a 3D printer, the possible synchronization error between motors and extruder can be neglected, because you have more variance in the plastic extrusion so a few us delay can not be noticed.

Lets assume Hobby scale as machines which move with speeds less then 10m/s and have precision less then 1um or similar, this is very vague, but applies to most machines at reasonable cost.

Timing between the cars is synced, so you can use it as it is already.

Jmjcoke, 2016-05-24:
Ok, 1um is small. I am good with 100-150 microns if I am lucky. I will let
you know. I have access to a API XD6 at work, I know it is overkill but it
will be fun to see how close it is.